lavolpecheprogramma/grbl-1-1h-servo #3: Servo PWM settings for travel

Dudley1001, 2021-02-26:
I have been trying to get the servo working with your firmware. I assumed the m3 s255 would get full servo travel but I have to go as high as S1000 to get a good range of travel. From what I see in the config.h the pwm is maxed at 255 but this does not appear to be the case. Any ideas? This is with a direct download of your code with no changes. Arduino uno/cnc shield V3

lavolpecheprogramma, 2021-02-26:
Hi,
you can try to tweak [this parameters](https://github.com/lavolpecheprogramma/grbl-1-1h-servo/blob/master/grbl/spindle_control.c#L24) .
Tell me if it can help you ;)

Dudley1001, 2021-02-26:
Ok, found the problem, the setting $30=1000 has to be changed to $30=255 in the settings. You could change it in defaults.h line44 to #define DEFAULT_SPINDLE_RPM_MAX 255 // rpm and that may help others figure it out.

lavolpecheprogramma, 2021-02-26:
Oh you are right! In my setting i have $30=80. But 255 can be a good default! thank you!

lavolpecheprogramma, 2021-02-26:
@Dudley1001 ;) https://github.com/lavolpecheprogramma/grbl-1-1h-servo/commit/4e4e84ec2b3fc0bdf940d062c7533c3f90f1893a

closed 2021-02-26 by lavolpecheprogramma