lavolpecheprogramma/grbl-1-1h-servo #4: Servo Travel Range

MarcAhoy, 2021-03-05:
Hello, Is there a way to make the servo travel the full amount, it only goes 90 degrees and back.

Thanks

Dudley1001, 2021-03-06:
Try changing the $30 value to 100. I recommended 255 but it was too much and 100 seems to work better

MarcAhoy, 2021-03-06:
I tried that but it still only goes 90°
What commands should I be using with $30 = 100
M3 S0 - to return ?
M3 S100 - for full turn?

MarcAhoy, 2021-03-06:
And does the motor type affect the rotation distance? 
I'm testing it with an MG996R, but am going to use it with an SG90 servo eventually

Dudley1001, 2021-03-06:
try setting it to $30=100 to a lower value and see what happens. It seems like the smaller you go the more rotation you get but it may limit the resolution as well. You can also change your M3 S to over the 100 but be careful as I stripped a servo when it went too far. Those are the 2 parameters that you can play with.  I only need about 90 degrees so it works well enough for me.  I have seen some random movement on the servo at the m3 S0 position, more like servo gitters.

MarcAhoy, 2021-03-06:
ok, I tried that and it works for what I need. 

Thanks for the help :D

closed 2021-03-06 by MarcAhoy