lavolpecheprogramma/grbl-1-1h-servo #7: Servo Motor on Z+ pin receives interference when Nema motors are working

murilobaliego, 2021-06-08:
Hi, I'm starting using this grbl, as I have a CoreXY plotter with Servo Motor on Z axis. 

The Servo Motor receives a kind if interference (it makes noise and moves quickly and short) when the main steppers are running. 

Firstly I thought it could be an issue in CNC Shield, so I replaced to one that was working in another machine with GRBL 0.9 and the problem persisted. I also replaced the Arduino Uno and still I get the same issue.

Any idea what could be the issue?

murilobaliego, 2021-06-08:
No, not yet, do you mean on PWM pin? Like a pull up or pull down resistor or a resistor between pin Z+ on CNC Shield and Orange wire of servo?
I saw some wire up with a capacitor on source pins of the servo, but this is to avoid drop of voltage when servo is turned on.

yusufiga, 2021-06-08:
Have you adjusted your steps/mm for your drivers? Default is I guess 250.000 which might be not correct for your case? That would mean checking values for $100,$101, $102

murilobaliego, 2021-06-08:
This I have adjusted but did not helped.

lavolpecheprogramma, 2021-06-08:
sorry but i don't know how can i help you. This happens to me too only when i move the plotter with jog but if i run a gcode, it works fine.
Tell me if you resolve it.

otienoelvis, 2022-05-01:
Please help, I have a SG92R servo,
which pins on the CNC shield v3 do I plug the wires ?

itsAliKing, 2024-11-23:
Hi Fount the solution in the GRBL Settings! You just need to decrease $30 value under 255 and the clicking random movement will gone. Oh GOD, it's such a big thing that I won't hear that clicking sound anymore. I Wouldn't get this behaviour on 0.9 version, because 0.9v doesn't even have $30, $31,$32 Options in the Settings. 
Please send feed backs if this solved the the clicking motion in your machines too.